roboterclubaachen/xpcc #345: scons fails to build for stm32f0xx devices.

lmoesch, 2018-02-12:
```
[xpcc\build\stm32f0_discovery\blink\main.o] UnicodeDecodeError : 'charmap' codec can't decode byte 0x81 in position 42: character maps to <undefined>
scons: building terminated because of errors.
```

Get this error for the stm32f0_discovery/blink example, aswell as my regular project code, which builds flawlessly for e.g. the stm32f746g_discovery or the f4 nucleo boards.

salkinium, 2018-02-12:
Position 42 in `stm32f0_discovery/blink/main.cpp` should be a newline value.
```
```

lmoesch, 2018-02-12:
Jeah, despite the error message, it does not seem to have anything todo with the main.cpp. When I change the build target, the error disappears, so I guess it has some problems with parsing/loading the device files.

Can you reproduce the error on Linux?

lmoesch, 2018-02-19:
The root of the problem were carelessly used "io.open" calls in the template.py, wich leads to enoding errors, when pythons default encoding ist not utf-8 (which it is unfortunately not under Windows /sigh). I temporarily fixed it, by specifying the encoding directly in all io.open calls. I guess the "correct" way would be trying to get the encoding from file or atleast try-catching all io.open calls so that useful error messages can be displayed.

salkinium, 2018-04-04:
Hm, this sucks, this was probably a side-effect of switching to Python 3, which IIRC has different behavior on file encodings.

lmoesch, 2018-04-05:
@salkinium I tested it with python2 aswell. This seems to be a python specific thing on Windows. (To be more specific the default encoding is **hardcoded** in the platformfile of the python installation)

@ekiwi 
`open(_path_, _mode_, encoding='utf-8')`
You should always specify the target files encoding, since there is no guarantee that your file has the "right" encoding. Since there is no good way to determine a files encoding, in my oppinion it is totally okay to say, that all files have to be utf-8 encoded.

salkinium, 2018-04-15:
Fixed in #349.

closed 2018-04-15 by salkinium