roboterclubaachen/xpcc #348: Building generated platform UART fails for STM32F030CC

lmoesch, 2018-04-09:
While including <xpcc/architecture.hpp> the building process fails for STM32F030CC with the following error:

> [...]\build\libxpcc\generated_platform\driver\uart\stm32\uart_hal_5_impl.hpp: In static member function 'static void xpcc::stm32::UsartHal5::enableInterruptVector(bool, uint32_t)':
[...]\build\libxpcc\generated_platform\driver\uart\stm32\uart_hal_5_impl.hpp:146:20: error: 'USART5_IRQn' was not declared in this scope
   NVIC_SetPriority(USART5_IRQn, priority);

This only happens for the 'CC' variant with more Uarts. E.g. 'C8' builds fine.

salkinium, 2018-04-09:
They probably have a shared interrupt, see this issue #88.
You may need to forward the functions by querying the status flags in that shared interrupt.

salkinium, 2018-04-09:
Hm, [the header file](https://github.com/modm-io/cmsis-header-stm32/blob/master/stm32f0xx/Include/stm32f030xc.h#L120) seems to not know about the `USART5_IRQn`, even though the peripheral is defined below. Maybe this is a bug?

`USART3_6_IRQn` is the shared interrupt, but that probably needs to be fixed in xpcc properly.

lmoesch, 2018-04-09:
Is the uart driver able to handle shared interrupts, or has this to be implemented aswell?

salkinium, 2018-04-09:
The UART driver will still generate separate calls to `NVIC_SetPriority` with values that aren’t defined. Probably the easiest way is to `#define USART3_IRQn USART3_6_IRQn` and then provide an `USART3_6_IRQ_Handler` based on the interrupt flags.

lmoesch, 2018-04-10:
Jeah, the implementation is a bit messy in the stm32 headerfile aswell, since there is `USART3_4_IRQn`  defined as `USART3_6_IRQn` and thatswhy only USART5/6 are crashing.

With USART5/6 replaced with USART3_6 the code compiles just fine. I just have the problem, that UART3-6 are somehow crashing, when more than one byte shall be sent (e.g. via `USART::write(*ptr,len)` or the IOWrapper stream), but this could also be a none related issue. I keep you updated :)

PR #351 (2018-04-19): Fixed shared UARTirq for stm32f030cc
Okay, this is my attempt to somehow fix the issue with shared IRQs on stm32 devices mentioned in #88 and #348 . The fix has two stages.

First new paramters are added to the `driver.xml`. A parameter called `shared_irq` containing the name of the shared UART (As far as I have seen it, this is either USART3_6 or USART3_8) and a parameter called `shared_irq_ids` containing a whitespace seperated list of the UART ids that are shared.

Than these new parameters are used to generate the appropriate code in the `uart.cpp`, `uart.hpp` and `uart_hal_impl.hpp`.

If  [... cut]
closed 2018-04-19 by salkinium